SolarEnergyRacers/solar_car_control_system #40: Change CAN Bitrate BMU

tomajon1, 2021-04-11:
Write Email to Prohelion

User Manual:
"COMMUNICATIONS
The CAN standard does not specify high-level message protocols. Prohelion devices use a
custom protocol, outlined in the communication specification document for each device.
By default, each device operates at 500 kbits/second and comes programmed from the factory
with a CAN base address that will allow it to work in unison with other Prohelion devices. The
CAN bit rate and base address can be set with the Windows BMS configuration software."

tomajon1, 2021-04-12:
Hi Jonas,
 
This functionality is not yet in Profinity, you will need to use the old Tritium tools.
 
Make sure you get the latest versions of BMS Viewer and CAN-Ethernet bridge configuration:
 
 BMS_Viewer_2_00.exe
 CanBridgeConfigInstaller_2_04.msi
 
 
If you run the BMS viewer software:
 
1.	Setup->BMS Configuration
2.	Change CAN Bitrate to desired rate:
 
3.	 
[image]

 
4.	Then: Transfer->Save to BMU
 
The BMU is now running on the new rate but you will have now lost communications so will need to change the rate on the CAN Bridge:
 
Go to CAN-Ethernet Bridge Configuration:
5.	Change Bit Rate
 
[image]

Hit:  Store
 
And you should now be communicating on the same bitrate.
 
Kind regards,
 
Anthony Prior
CEO
M: +61 411 098 131

andreaskuster, 2021-04-25:
@Damian:

(1) read datasheet
(2) set bitrate to 125k
(3) get it working :)
- possible solution: twisted-pair cable
- oscillator debugging

tomajon1, 2021-07-13:
CAN-Ethernet Bridge Configuration does not work on Windows 7

andreaskuster, 2021-09-06:
Solved & PR created.

closed 2021-09-06 by andreaskuster