SolarEnergyRacers/solar_car_control_system #166: Integrate HC-12 radio module

KarlSchu, 2022-02-15:


andreaskuster, 2022-02-15:
@KarlSchu 
A little present for you:
(1) UART Tx/Rx must be twisted on both sides (i.e. Rx <-> Tx and vice versa)
(2) The SET pin must be floating
(3) It is actually pretty easy to use multiple Serial instances in parallel, see code below (for the receiver)
```
#define RXD2 16
#define TXD2 17

void setup() {
  // Note the format for setting a serial port is as follows: Serial2.begin(baud-rate, protocol, RX pin, TX pin);
  Serial.begin(115200);
  
  Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
  
  Serial.println("Serial TX0 is on pin: "+String(TX));
  Serial.println("Serial RX0 is on pin: "+String(RX));

  Serial.println("Serial TX2 is on pin: "+String(TXD2));
  Serial.println("Serial RX2 is on pin: "+String(RXD2));

}

bool payload;
void loop() {
  payload = false;
  while (Serial2.available()){
    Serial.print(char(Serial2.read()));
    payload = true;
  }

  if(!payload){
      Serial.println("Nothing to process");
  } else {
      Serial.println();
  }
  delay(5000);
}
````

andreaskuster, 2022-02-20:
Notes: 
- continuous 5V supply ??
- capacitor on tx side